makermoekoe/ESP32Picoremote #1: Latching Circuit

mekesim, 2021-04-22:
Hi Sir,
thanks for your sharing such excellent project.
I have a question about the latching circuit. How does it works, could you explain it please? I can find some basic work flow of it.

- while USB insert, EN_LDO high, it will enable the LDO.
- Button 1 or Button 2 pushed, EN_LDO high, it will enable the LDO.
- LATCH low, power off LDO, right?

Why there are tow button, how about just one button?

makermoekoe, 2021-04-23:
Hey, thank you!

Basically you have mentioned all the important features. It is as simple as controlling the enable pin of the LDO. If it's high, the LDO will deliver power to the MCU and all the devices which are connected to the output of the LDO. If not, it won't. As you have stated, the enable pin can be controlled via both buttons and a specific GPIO of the MCU. 

Basic scenario is:
1. Press one of the two buttons (for other designs this could be even more buttons)
2. Hold the GPIO high (in fact the enable pin will be held high) so that the LDO will still deliver power once the button is released
3. If your program is done, you can just set the GPIO low and the MCU will shut down

For the newest version of the schematic and PCB I have added the enable control via USB voltage. Once it is connected to 5V it will pull the enable pin high. This won't affect the battery because once a USB voltage is connected the whole board (or in detail: the VCC line) is powered via the 5V USB voltage (through the diode D3). And, of course, the battery will be charged in this state.

Hope this clarifies it a bit more. If you still have questions feel free to ask!

mekesim, 2021-04-25:
Hi Sir, 
thanks for your explain. I'm clear now.

An other question.
U3 is able to do reverse voltage protection,  why do you add Q1? Any other consideration?

makermoekoe, 2021-04-27:
Yes, you're probably right. I mixed up two circuits and haven't noticed that I've doubled the polarity protection...
Will fix it in the next revision!

closed 2021-04-27 by makermoekoe