makermoekoe/ESP32Picoremote #2: Auto RESET USB

custanator, 2021-04-26:
Hi! 
Thanks for posting a good project!
On Instagram, you wrote about the problem of automatic firmware management via USB
In your circuit, there are no resistors in a pair of transistors.
They are present in the official scheme of NodeMСU.
Maybe this is the problem?

[image]
[image]

makermoekoe, 2021-04-27:
The BC847s is a dual npn transistor which will need two resistors for sure. The MUN5214DW1 has exactly the same pinout but it includes a simple bias resistor network. Therefore, the two resistors on the base are not necessary in this case. You can check the datasheet [here](https://www.onsemi.com/pdf/datasheet/dtc114yd-d.pdf).

The problem which I've mentioned on Instagram (which results in this huge pcb creature) was just a wrong capacitor value for the enable pin of the ESP32. It needs 10uF there.

closed 2021-04-27 by makermoekoe

custanator, 2021-04-27:
All clear. Thank you!