makermoekoe/ESP32Picoremote #3: What about PMIC?

custanator, 2021-04-26:
Have you thought about replacing all the power supply controls with a microcircuit PMIC (Power Management IC)?

Example: AXP173
It already contains:
- Power Patch (Power selection BAT/USB)
- 2 DC/DC
- 4 LDO
- Charge control
- Gas Gauge (how much charge is left in the battery)
- Button On/Off
- QFN-32

Using PMIС would greatly simplify the circuit :)

mekesim, 2021-04-27:
So powerful!

custanator, 2021-04-27:
Other variants

- AMS AS3701 –– Power Path, 1 DC/DC, 2 LDO, WLCSP-20, charger, GPIO and button
- Maxim MAX20353 –– Power Path, 3 DC/DC, 2 LDO, WLP-56, charger, gas gauge, GPIO, button
- Dialog DA9070 –– Power Path, 1 DC/DC, 3 LDO, WLCSP-42, charger, gas gauge (ADC), watchdog, button
- TI TPS6572x –– Power Path, 1 DC/DC, 1 LDO, WQFN-32, DSBGA-25, charger, GPIO, button
- Qorvo (Active Semi) ACT81460 –– Power Path, 4 DC/DC, 3 LDO, 1 HVB, WLCSP-49, charger, 4 GPIO, button
- Microchip MCP73871 — Power Path, QFN-20, charger.

makermoekoe, 2021-04-27:
Wow, thanks for this nice table of PMICs! I will check them all later the day. 

But what about power consumption in shutdown state? At first sight it seems they're all using at least 25uA of power... Nevertheless, this would combine everything in just one chip and make the PCB more beautiful

custanator, 2021-04-27:
Yes, X-Powers AXP173 Off mode - 27uA. IMHO, its normal =)

makermoekoe, 2021-04-27:
Well, it is definitely low power! But the circuit in this project consumes just about 3uA which is related to the battery protection only.
I will have a deeper look on such a chip for upcoming projects 👌🏻

closed 2021-04-27 by makermoekoe