makermoekoe/ESP32Picoremote #6: Two question (USB-C and Battery)

Theagainmen, 2022-05-03:
Hello all,

Thanks for creating such an awesome project! I really learn a lot from looking at the schematics and code. 

For a project I am working on I wanted to use the USB-C part from your schematics. However I find many different variants of the USB-C, some with just 6 pins, up to 32 with many options in between. In your schematic you use one that only shows 7 pins, do you maybe have the exact component name of the USB-C you have used?

About the battery part I had also a question, you use the XB5353A (protection) and MCP73831 (charge) to manage the LiPo battery, I'm wondering if these IC's allow the battery to be discharged at the same time it's charged? The TP4056 for example does not allow this, since this can result in an overcharge failure detection, but I'm unable to find anything about it in the IC datasheets you have used. 

Thanks for any help :)

PathToLife, 2022-05-03:
Regarding the second part of this issue
- https://electronics.stackexchange.com/questions/539946/recharging-a-lipo-battery-while-using-it

Charger ICs that can support a load while still charging correctly exist, but your MCP73831 isn't one of them. The [MCP73871](https://ww1.microchip.com/downloads/en/DeviceDoc/MCP73871-Data-Sheet-20002090E.pdf) is; it supports load sharing and can supply a load while charging the battery.

Theagainmen, 2022-05-04:
> Regarding the second part of this issue
>
> * https://electronics.stackexchange.com/questions/539946/recharging-a-lipo-battery-while-using-it
>
> Charger ICs that can support a load while still charging [...]

Thanks! I'll look more into this. Do you maybe know how to even find IC's like this? Like what search engine/site do you go to and what to type in? Also to get some IC's that are actually used by more people and affordable :)

Theagainmen, 2022-05-04:
> About USB C... Yes, they might have 32 pins, but they are doubled. As the C standard has two rows, with the same function of pins, top and bottom pins should be connected. Say DN1 and DN2 are basically the same but on opposite sides.
> 
> In basic usage, you need DP, DN, CC, and Power pins. Do not be confused about CC1 and CC2, DN1 and DN2, and similar. You just have to connect those 1 and 2's. CC pins are a source to sink pins. Just check some datasheets.

Thanks, I've been checking some datasheets but I'm still not 100% whether they are compatible. 

If I just connect these pins in my KiCad project: `VUSB, CC1, CC2, D1-, D1+, SHIELD, GND` on this [USB-C (click)](https://datasheet.lcsc.com/lcsc/2103051833_Jing-Extension-of-the-Electronic-Co--918-418K2023S40001_C167321.pdf ), and leave the rest unconnected, will that work?

ITstreet1, 2022-05-04:
You only need VUSB, GND, D- and D+ if you are using it with a standard USB (PC or regular charger). Let's say for uploading a software. If you are using it with a charger that uses CC lines you must connect them, too. I would connect both CCs with a 5.1K to a GND just in case. The rest pins you can leave unconnected.

You can check the schematic here for the Picoremote.

As for the MCP73831, it is so easy to type in Google. 

https://eu.mouser.com/new/microchip/microchip-mcp73831-2-controllers/
https://www.digikey.com/en/products/detail/microchip-technology/MCP73831T-2ACI-OT/964301

...

Theagainmen, 2022-05-04:
Ok thank you @ITstreet1. I found this [USB-C](https://datasheet.lcsc.com/lcsc/2103051833_Jing-Extension-of-the-Electronic-Co--918-418K2023S40001_C167321.pdf ) and I'm connecting it like this:
[image]
I'm pretty sure this works, but if you see anything odd, please let me know!

> As for the MCP73831, it is so easy to type in Google.

As for this I was more talking about finding IC's **like this**. Like what tool or engine can I use to find IC's that charge LiPo and have X and Y specs. I know stuff like Mouser, but when I type in LiPo charger I get so many options, and many of them are not what I'm looking for.

ITstreet1, 2022-05-05:
Schematic look ok to me. 

> As for this I was more talking about finding IC's like this. Like what tool or engine can I use to find IC's that charge LiPo and have X and Y specs. I know stuff like Mouser, but when I type in LiPo charger I get so many options, and many of them are not what I'm looking for.

I don't know if such an online tool exists.

closed 2022-09-07 by Theagainmen